tessel/reach-wg #17: Write a guide to loading our `firmware-esp32/` code onto an ESP32 Thing.

tcr, 2017-02-18:
This should be straightforward; we have code in`firmware-esp32/` and we want an instructions file at `firmware-esp32/README.md` which describes how to load code onto the device.

This may increase in complexity over time, but for now can mostly replicate the instructions at <https://github.com/espressif/esp-idf-template>, e.g:

```
## Getting started

* Plug in your ESP32 Thing.
* Run the makefile.
* Load it onto the device.
* Expect $THIS_THING to happen.

### Troubleshooting

Possible errors with drivers? etc. Follow these instructions: $INSTRUCTIONS_URL
```

tcr, 2017-02-18:
Great jaerb there, Homestar. @EvanSimpson

closed 2017-02-18 by tcr

Frijol, 2018-06-25:
closed in https://github.com/tessel/reach-wg/pull/19