tessel/reach-wg #18: Create an ESP32 demo talking SPI to an accelerometer.

tcr, 2017-02-18:
We want the ESP32 demo at `firmware-esp32/` to be capable of reading data from an accel-mma84 module over I2C. The MVP would be reading acceleration data and turning on lights when e.g. the board is flipped over.

Here is some example code for talking to the MMA8452q accelerometer over I2C: <https://github.com/tessel/tessel-rust/blob/master/accel-mma84/src/lib.rs#L46>

We should be able to test this independently using a logic analyzer to see that the I2C driver is correct.

nodebotanist, 2017-02-23:
I could do this, but I don't know if I have that accelerometer. Let me check my stash before I commit to anything.

EvanSimpson, 2017-02-24:
If I can find the time this weekend I can take a stab at this - just found an old accelerometer module. 

As an aside I also just put up some code I'm working on which has an esp32 successfully talking to the Tessel IR module over SPI -> [here](https://github.com/EvanSimpson/esp-remote). My next steps for that one are getting a basic GATT server set up and sending IR commands through lightBlue on my laptop.